Embedded-AMS/EmbeddedProto #2: Trouble getting venv to work on Ubuntu 18.04

Michaelhobo, 2020-08-07:
Setup steps:
Installed python3, pip, venv, protobuf, git, cmake, etc.
Ran setup.sh script

Problem:
All paths in protoc-gen-eams are relative. When generating protos from a different directory, these paths all fail to resolve.  

Solution:
Add $(dirname $0) in front of both ./venv/bin/python and generator/protoc-gen-eams.py

sgerbino, 2021-09-15:
It would be nice to leverage `protoc-gen-eams` from an external directory (e.g. when utilizing the CMake macro `protobuf_generate`).

BartHertog, 2021-09-17:
@sgerbino It is likely we will pick this up with a next minor release.

BartHertog, 2021-09-28:
@Michaelhobo Dirname has been implemented. Sorry for this long delay in this simple ticket.

BartHertog, 2021-09-28:
@sgerbino Implemented the dirname solution. It is in develop, I have not tested it with `protobuf_generate`, will you let me know it it works for you?

BartHertog, 2021-10-04:
@Michaelhobo @sgerbino It is now in version 2.3.2

closed 2021-10-04 by BartHertog

sgerbino, 2021-10-05:
I'm currently used a hacked up version of your fixes in 2.x and 3.0 release here: https://github.com/koinos/EmbeddedProto

I won't be able to test it out until the 3.0 release @BartHertog

Thanks for this update, I will let you know as soon as it's available in the 3.0 version

BartHertog, 2021-10-05:
@sgerbino I will merge the current 2.x state in to the release candidate branch probably in the coming week.