Embedded-AMS/EmbeddedProto #3: Generated code not compiling with gcc-arm-none-eabi-10-2020-q2-preview

Michaelhobo, 2020-08-07:
Different issues on two different toolchains:
ARM GCC 10: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads
Linux GCC 10.1, installed from Ubuntu's gcc-10 package.

python 3.6.9

With ARM GCC 10, I'm getting errors like this:
EmbeddedProto/src/RepeatedField.h:54:34: error: expected unqualified-id before '=' token
   54 |     static constexpr bool PACKED = !std::is_base_of<MessageInterface, DATA_TYPE>::value;

...

EmbeddedProto/src/RepeatedField.h:137:12: error: expected ')' before '__attribute__'
  137 |         if(PACKED)
      |           ~^
      |            )

Maybe there are some issues with attributes with this toolchain? I'm not sure.

With Linux GCC 10.1, I'm getting errors like this:
proto/example.h:54:16: error: 'B' does not name a type
   54 |   inline const B& b() const { return b_; }

This is because the generated code generates message A first, without forward declaring message B.

Proto file:
`syntax = "proto3";

message A {
  B b = 1;
}

message B {
  int32 val = 1;
}
`

Michaelhobo, 2020-08-08:
For the GCC 10.1 error, I confirmed - I can compile if I reverse the order of message A and message B in the proto. This shouldn't be necessary though.

Michaelhobo, 2020-08-08:
I moved the proto ordering errors into a new issue, since that seems like a pretty separate issue

BartHertog, 2020-08-11:
With regards to the compiler error we I am unable to reproduce them at this moment. Can you eleborate on the proto file used when this error occured. The example file you list does not hold any repeated fields.

Michaelhobo, 2020-08-12:
After rearranging the messages and replacing the broken header guard with #pragma once, I now can't reproduce this. But, I also deleted the proto file that I used to generate this, so I don't know if it's a problem or not still.

I'll close it for now, and I will open this again if it becomes a problem.

closed 2020-08-12 by Michaelhobo