Embedded-AMS/EmbeddedProto #4: Order of proto messages in .proto matters

Michaelhobo, 2020-08-08:
The order of proto message definition in the .proto file matters. If a proto message is defined after it is used, the generated code fails to compile. I have seen 2 concrete issues so far:

1. No forward declarations happen
2. template arguments for fields are omitted

## 1. No forward declarations happen

```proto
message A {
  B b = 1;
}

message B {
  int32 val = 1;
}
```

This fails because B hasn't been declared yet, and class A references an undefined label B.

## 2. template arguments for fields are omitted

```proto
message D {
  C c = 1;
}

message C {
  repeated int32 list = 1;
}
```

This fails because of #1, but the template arguments for C are not generated. Instead, this gets generated:

```cpp
class D final: public ::EmbeddedProto::MessageInterface
{
  public:
    D() :
        c_()
    {

    };
    ~D() override = default;

    enum class id
    {
      NOT_SET = 0,
      C = 1
    };

    inline const C& c() const { return c_; }
    inline void clear_c() { c_.clear(); }
    inline void set_c(const C& value) { c_ = value; }
    inline void set_c(const C&& value) { c_ = value; }
    inline C& mutable_c() { return c_; }
    inline const C& get_c() const { return c_; }

    ::EmbeddedProto::Error serialize(::EmbeddedProto::WriteBufferInterface& buffer) const final
    {
      ::EmbeddedProto::Error return_value = ::EmbeddedProto::Error::NO_ERRORS;

      if(::EmbeddedProto::Error::NO_ERRORS == return_value)
      {
        return_value = c_.serialize_with_id(static_cast<uint32_t>(id::C), buffer);
      }
       

      return return_value;
    };

    ::EmbeddedProto::Error deserialize(::EmbeddedProto::ReadBufferInterface& buffer) final
    {
      ::EmbeddedProto::Error return_value = ::EmbeddedProto::Error::NO_ERRORS;
      ::EmbeddedProto::WireFormatter::WireType wire_type;
      uint32_t id_number = 0;

      ::EmbeddedProto::Error tag_value = ::EmbeddedProto::WireFormatter::DeserializeTag(buffer, wire_type, id_number);
      while((::EmbeddedProto::Error::NO_ERRORS == return_value) && (::EmbeddedProto::Error::NO_ERRORS == tag_value))
      {
        switch(id_number)
        {
          case static_cast<uint32_t>(id::C):
          {
            if(::EmbeddedProto::WireFormatter::WireType::LENGTH_DELIMITED == wire_type)
            {
              uint32_t size;
              return_value = ::EmbeddedProto::WireFormatter::DeserializeVarint(buffer, size);
              ::EmbeddedProto::ReadBufferSection bufferSection(buffer, size);
              if(::EmbeddedProto::Error::NO_ERRORS == return_value)
              {
                return_value = mutable_c().deserialize(bufferSection);
              }
            }
            else
            {
              // Wire type does not match field.
              return_value = ::EmbeddedProto::Error::INVALID_WIRETYPE;
            } 
            break;
          }

          default:
            break;
        }
        
        if(::EmbeddedProto::Error::NO_ERRORS == return_value)
        {
            // Read the next tag.
            tag_value = ::EmbeddedProto::WireFormatter::DeserializeTag(buffer, wire_type, id_number);
        }
      }

      // When an error was detect while reading the tag but no other errors where found, set it in the return value.
      if((::EmbeddedProto::Error::NO_ERRORS == return_value)
         && (::EmbeddedProto::Error::NO_ERRORS != tag_value)
         && (::EmbeddedProto::Error::END_OF_BUFFER != tag_value)) // The end of the buffer is not an array in this case.
      {
        return_value = tag_value;
      }

      return return_value;
    };

    void clear() final
    {
      clear_c();
    }

  private:

    C c_;

};
```

Note that C should be C<c_LIST_LENGTH> and the template should be passed through class D.

BartHertog, 2020-08-11:
Dear Micheal,

This issue we are able to reproduce. Fixing it by adding the forward declarations is however not the solution. As you are probably aware of forward declarations are only valid for pointer and reference variables. However in embedded proto we do not use dynamic memory allocation (one of it's most important premises) meaning that the instance of message b is thus defined in message c. Regular protobuf does work by dynamically allocating message b and storing the pointer in message c. For now the only solution I can offer is to define the messages in order.

Best regars,

Bart

Michaelhobo, 2020-08-11:
Is it not possible to do that reordering in the generated code? It seems reasonable to check the order of a class's dependencies, and make sure that the class is defined after all its dependencies.

I'm fine with defining messages in order as a workaround, as long as there's a solution on the roadmap for fixing this in the long run.

BartHertog, 2020-08-12:
A ticket has been made:
"On github it was pointed out that the order of the messages in the proto file does not matter. It does however matter in C++ the way we use it.
Please see if we can sort the messages in python to see if we can solve this. If a circular dependency is found an error should be thrown."

Michaelhobo, 2020-08-26:
Any update on this?

BartHertog, 2020-08-31:
Sort of, we are working on a different ticket, to support nested message definitions. When this is implemented it is most likely easier to order the messages into a logical order required for this ticket.

Michaelhobo, 2020-08-31:
I see, understood.

BartHertog, 2020-09-13:
Hi Michalel,
I've been trying to get a solution working for this problem but so far I have been running into limitations of the generator. I am contemplating a new design for the generator code. Please bare with me. 
Best regards,
Bart

Michaelhobo, 2020-09-13:
No problem, thank you for sticking with the problem.

Michaelhobo, 2020-10-19:
Since there's an order that should work (after all, a human can follow a mechanical set of rules and do it manually), do you think it's reasonable to rearrange the message_type repeated field here in a preprocessing step?
https://sourcegraph.com/github.com/protocolbuffers/protobuf@ce66f6047db44df3234bccd6fcf468bccc14a760/-/blob/python/compatibility_tests/v2.5.0/protos/src/proto/google/protobuf/descriptor.proto#L69

For example, you could put the DescriptorProto messages into a dependency graph as an intermediary step, then regenerate the message_type list with the contents of the graph.

BartHertog, 2021-11-16:
Hello @Michaelhobo,

My apologies for the long delay but it has finally been fixed.  A sorting algorithms has been included which sorts the messages in the order of inclusion. You can find this on the develop branch.

Let me know if you have any remarks.

Best regards,

Bart

closed 2021-11-16 by BartHertog