Embedded-AMS/EmbeddedProto #12: Macro collision with PACKED in nrf52 SDK

Michaelhobo, 2020-08-14:
I am using the nrf52 SDK, and unfortunately, they define a PACKED macro deep in their library, which is causing a collision with RepeatedField.h's static constexpr bool PACKED. 
I realize this isn't a problem with your library, and I can also change it locally in my copy of EmbeddedProto. But, I would like if EmbeddedProto could choose a different name for this variable.

Maybe consider Google's constants naming convention, which would make this `kPacked`? https://google.github.io/styleguide/cppguide.html#Constant_Names

BartHertog, 2020-08-15:
So as I understand correctly they are defining it publicly?

Michaelhobo, 2020-08-15:
Yes, they are... this is why I don't like C and macros :( 

They have:
#define PACKED __attribute__((packed))

I do think this is a relatively common macro. I see a couple people suggesting writing this macro or variants of this macro to make things "easier"

BartHertog, 2020-08-17:
Done

closed 2020-08-17 by BartHertog