Embedded-AMS/EmbeddedProto #13: undefined reference to operator delete(void*, unsigned int)

Michaelhobo, 2020-08-14:
I'm building an image with gcc-arm-none-eabi using the most recent develop branch with some of the fixed issues. It now compiles, but it doesn't link properly, because there are a lot of calls to operator delete. Is there some memory allocation happening?

```bash
/toolchains/gcc-arm-none-eabi-10-2020-q2-preview/arm-none-eabi/bin/ld: _build.DEV/_.cpp.o: in function `a::G::~G()':
_.h:56: undefined reference to `operator delete(void*, unsigned int)'
/toolchains/gcc-arm-none-eabi-10-2020-q2-preview/arm-none-eabi/bin/ld: _build.DEV/_.cpp.o: in function `a::C<17ul>::~C()':
_.h:214: undefined reference to `operator delete(void*, unsigned int)'
/toolchains/gcc-arm-none-eabi-10-2020-q2-preview/arm-none-eabi/bin/ld: _build.DEV/_.cpp.o: in function `EmbeddedProto::RepeatedFieldFixedSize<a::G, 17ul>::~RepeatedFieldFixedSize()':
/EmbeddedProto/src/RepeatedFieldFixedSize.h:64: undefined reference to `operator delete(void*, unsigned int)'
/toolchains/gcc-arm-none-eabi-10-2020-q2-preview/arm-none-eabi/bin/ld: _build.DEV/_.cpp.o: in function `EmbeddedProto::MessageSizeCalculator::~MessageSizeCalculator()':
/EmbeddedProto/src/MessageSizeCalculator.h:54: undefined reference to `operator delete(void*, unsigned int)'
/toolchains/gcc-arm-none-eabi-10-2020-q2-preview/arm-none-eabi/bin/ld: _build.DEV/Fields.cpp.o: in function `EmbeddedProto::int32::~int32()':
/EmbeddedProto/src/Fields.h:123: undefined reference to `operator delete(void*, unsigned int)'
/toolchains/gcc-arm-none-eabi-10-2020-q2-preview/arm-none-eabi/bin/ld: _build.DEV/ReadBufferSection.cpp.o:/EmbeddedProto/src/ReadBufferSection.h:63: more undefined references to `operator delete(void*, unsigned int)' follow
collect2: error: ld returned 1 exit status
```

Michaelhobo, 2020-08-14:
Specifically, I think this happens when using a message that contains a repeated field.

```proto
message G {
  int32 a = 1;
}

message C {
  repeated G gs = 1;
}
```

If I create a C<17>, I get the linker error. If I just create G messages and store them elsewhere, it links properly.

BartHertog, 2020-08-15:
We will be working on this. It is really inserting besides the technical things. May I ask you why and how are you using dynamic memory allocation in your embedded application? 
As I mentioned before Embedded Proto was mainly designed not to use dynamic allocation and from you mentioning delete I suspect you do. So I am eager to learn what your motivation and situation is to do so. Are you using an OS for instance which deals with the memory issues on micro controllers?

BartHertog, 2020-08-15:
I do not seem to be able to reproduce this. I've used the makefile example, this has a repeated message and a oneof. Those are the only special things. 
These is no manual calling of delete in the code. The default destructors are defined. I have to look into why it is asking for "delete(void*, unsigned int)" in particular.

Michaelhobo, 2020-08-15:
Are you using a Cortex-M ARM toolchain to reproduce? The toolchain doesn't define operator delete by default.
>  I've used the makefile example,
Are you talking about this example? https://github.com/Embedded-AMS/EmbeddedProto_Example_Makefile/tree/master/proto

I don't see a repeated message in it?

Michaelhobo, 2020-08-16:
It turns out this has nothing to do with the proto contents (repeated, int, oneof, etc) - I was able to reproduce it no matter what fields the message contained. I think there's some interaction with the base class destructors that I don't fully understand yet.

I got this to link by getting rid of all the virtual destructors in all the base classes - I removed the destructors in MessageInterface, Field, WriteBufferInterface, etc. But, this may cause problems with derived class destructors not getting called.

Michaelhobo, 2020-08-16:
Here's a minimal example that fails for me:
```c++
class P {
 public:
  void Do(const M& msg) {
    std::array<uint8_t, 20> buf;
    pbtool::WiteBuffer w(buf);
    msg.serialize(w);
  }
};
int main() {
  M m;
  m.set_int(1);
  m.set_bool(true);
  P p;
  p.Do(m);
}
```

Here's most of my WriteBuffer implementation. Only difference is I use absl::Span instead of std::array in my real implementation, but that requires including a separate lib.

```c++
class WriteBuffer : public EmbeddedProto::WriteBufferInterface {
 public:
  WriteBuffer(std::array<uint8_t, 20> data) : data_(data), index_(0) {}
  ~WriteBuffer() = default;
  void clear() override { index_ = 0; }
  inline uint32_t get_size() const override { return index_; }
  inline uint32_t get_max_size() const override { return data_.size(); }
  inline uint32_t get_available_size() const override {
    return data_.size() - index_;
  }
  bool push(const uint8_t* bytes, const uint32_t length) override {
    if (index_ + length > data_.size()) {
      return false;
    }
    std::copy(bytes, bytes + length, data_.begin() + index_);
    index_ += length;
    return true;
  }
  bool push(const uint8_t byte) override { return push(&byte, 1); }

 private:
 std::array<uint8_t, 20> data_;
  size_t index_;
};
```

This fails to link with this (redacted):

```bash
"make" -s --no-print-directory -f "nrf_sdk/components/toolchain/gcc/dump.mk" VARIABLE=CONTENT_TO_DUMP > _build.DEV/nrf52832_xxaa.in
'toolchains/gcc-arm-none-eabi-10-2020-q2-preview/bin/arm-none-eabi-gcc' -Os -g3 -mthumb -mabi=aapcs -L nrf_sdk/components/toolchain/gcc -Texample.ld -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Wl,--gc-sections --specs=nano.specs @_build.DEV/nrf52832_xxaa.in -Wl,-Map=_build.DEV/nrf52832_xxaa.map -o _build.DEV/nrf52832_xxaa.out
'toolchains/gcc-arm-none-eabi-10-2020-q2-preview/bin/arm-none-eabi-size' _build.DEV/nrf52832_xxaa.out
toolchains/gcc-arm-none-eabi-10-2020-q2-preview/arm-none-eabi/bin/ld: _build.DEV/main.cpp.o: in function `pbtool::WriteBuffer::~WriteBuffer()':
common/pbtool.h:58: undefined reference to `operator delete(void*, unsigned int)'
toolchains/gcc-arm-none-eabi-10-2020-q2-preview/arm-none-eabi/bin/ld: _build.DEV/nrf52832_xxaa/main.cpp.o: in function `M::~M()':
proto/msg.h:348: undefined reference to `operator delete(void*, unsigned int)'
toolchains/gcc-arm-none-eabi-10-2020-q2-preview/arm-none-eabi/bin/ld: _build.DEV/nrf52832_xxaa/Fields.cpp.o: in function `EmbeddedProto::int32::~int32()':
EmbeddedProto/src/Fields.h:123: undefined reference to `operator delete(void*, unsigned int)'
toolchains/gcc-arm-none-eabi-10-2020-q2-preview/arm-none-eabi/bin/ld: _build.DEV/nrf52832_xxaa/Fields.cpp.o: in function `EmbeddedProto::boolean::~boolean()':
EmbeddedProto/src/Fields.h:207: undefined reference to `operator delete(void*, unsigned int)'
toolchains/gcc-arm-none-eabi-10-2020-q2-preview/arm-none-eabi/bin/ld: _build.DEV/nrf52832_xxaa/Fields.cpp.o: in function `EmbeddedProto::MessageSizeCalculator::~MessageSizeCalculator()':
EmbeddedProto/src/MessageSizeCalculator.h:54: undefined reference to `operator delete(void*, unsigned int)'
collect2: error: ld returned 1 exit status
```

Michaelhobo, 2020-08-16:
I found the issue! I needed to pass -lstdc++ to my linker. https://bytes.com/topic/c/answers/877895-undefined-reference-operator-new-operator-delete

closed 2020-08-16 by Michaelhobo