Embedded-AMS/EmbeddedProto #14: repeated string is interpreted as non-repeated string

inada-s, 2020-09-10:
Thank you for publishing an easy-to-use library.
I had a problem when trying to use `repeated string`.

## Tested on
- EmbeddedProto: develop fb53b82721658b7aaaf868a5557d56cffa083532
- OS: Windows10 Home / wsl

### .proto
```
message repeated_fields 
{
  uint32 x            = 1;
  repeated uint32 y   = 2;
  uint32 z            = 3;

  string str   = 4;
  repeated string rep_str  = 5;
}
```

### genereated code
```
    EmbeddedProto::uint32 x_;
    ::EmbeddedProto::RepeatedFieldFixedSize<EmbeddedProto::uint32, y_LENGTH> y_;
    EmbeddedProto::uint32 z_;
    ::EmbeddedProto::FieldString<str_LENGTH> str_;
    ::EmbeddedProto::FieldString<rep_str_LENGTH> rep_str_;
```

### expected result
```
    ::EmbeddedProto::RepeatedFieldFixedSize<::EmbeddedProto::FieldString<rep_str_xxx_LENGTH>,rep_str_LENGTH> rep_str_;
```

BartHertog, 2020-09-13:
Hello inada-s,

Thank you for reporting this issue. We will be looking into this problem.

Best regards,

Bart

BartHertog, 2020-10-27:
Hello inada-s,
A new update has been made on the develop branch. This should fix the repeated string problem. 
Best regards,
Bart

closed 2021-01-13 by BartHertog