Embedded-AMS/EmbeddedProto #16: Optional fields support

DenDasus, 2020-11-14:
Hello!

Please, add optional fields support in proto3 syntax. This implemented by **--experimental_allow_proto3_optional** protoc flag. This feature will be added to protoc soon, but now it is experimental.

Error string: _Api.proto: is a proto3 file that contains optional fields, but code generator protoc-gen-eams hasn't been updated to support optional fields in proto3. Please ask the owner of this code generator to support proto3 optional._

BartHertog, 2021-01-13:
Hello DanDasus,

I will look into this. It is an extensive request as far as I can see now. All fields should get a boolean and function to get the boolean which indicates if the value has been set. However I do have felt the need for this from time to time when using proto3.

BartHertog, 2021-08-09:
Work on this ticket has started.

BartHertog, 2021-09-07:
The feature has been implemented. As of Embedded Proto version 2.3.0 it is possible to use the optional keyword in proto files. Please remember to use the proper protoc version.

closed 2021-09-07 by BartHertog