Embedded-AMS/EmbeddedProto #19: String fields use an undefined setter method.

altroniq, 2021-04-09:
Hello,

I've found a potential issue with the auto-generated header code when it comes to message fields of protobuf type `string` (which becomes type `::EmbeddedProto::FieldString<_LENGTH>`).

To start, I am seeing this on tag `2.1.0` and on the latest (as of the time of this writing) `master`.

Here is the issue I'm seeing. Let's say you define a protobuf message like so:

```
message FooMessage {
  string foo_string = 1;
  uint64 foo_int = 2;
}
```

In the generated header a portion of the class `FooMessage` will look like this:

```
template<uint32_t foo_string_LENGTH>
class FooMessage final: public ::EmbeddedProto::MessageInterface
{
  public:
    FooMessage() :
        foo_string_(),
        foo_int_(0U)
    {

    };
    ~FooMessage() override = default;

    FooMessage& operator=(const FooMessage& rhs)
    {
      set_foo_string(rhs.get_foo_string());
      set_foo_int(rhs.get_foo_int());
      return *this;
    }

    inline void clear_foo_string() { foo_string_.clear(); }
    inline ::EmbeddedProto::FieldString<foo_string_LENGTH>& mutable_foo_string() { return foo_string_; }
    inline const ::EmbeddedProto::FieldString<foo_string_LENGTH>& foo_string() const { return foo_string_; }
    inline const char* get_foo_string() const { return foo_string_.get_const(); }

    inline void clear_foo_int() { foo_int_.clear(); }
    inline void set_foo_int(const EmbeddedProto::uint64& value) { foo_int_ = value; }
    inline void set_foo_int(const EmbeddedProto::uint64&& value) { foo_int_ = value; }
    inline EmbeddedProto::uint64& mutable_foo_int() { return foo_int_; }
    inline const EmbeddedProto::uint64& get_foo_int() const { return foo_int_; }
    inline EmbeddedProto::uint64::FIELD_TYPE foo_int() const { return foo_int_.get(); }

...

```

The issue is in the overload of the assignment operator. The method `set_foo_string()` is not defined and if this operator is used, a compilation error is generated.

```
    FooMessage& operator=(const FooMessage& rhs)
    {
      set_foo_string(rhs.get_foo_string());  // <----- THIS IS NOT DEFINED ANYWHERE
      set_foo_int(rhs.get_foo_int());
      return *this;
    }
```

After looking at the implementation of `EmbeddedProto::FieldString`, I would propose the following fix (which appears to compile):

```
    FooMessage& operator=(const FooMessage& rhs)
    {
      foo_string_ = rhs.get_foo_string();  // <----- CHANGE TO THIS
      set_foo_int(rhs.get_foo_int());
      return *this;
    }
```

This appears to do the right thing, but I might also be missing something. 🤷🏽‍♂️

Thank you for taking the time to listen to this and even more thanks for creating EmbeddedProto!

-a

BartHertog, 2021-04-10:
Hello Altroniq,

Thank you for writing in. We will be looking at this issue.

Best regards,

Bart

BartHertog, 2021-04-11:
Hello Altroniq,
A fix has been implemented in develop. A discrepancy was found between the String/Bytes fields and the other regular fields. This was corrected in this ticket. It however created a breaking change. The return types of two type of get functions where switched around. So if you where using these you might have to change them as well.
Best regards,
Bart

closed 2021-07-22 by BartHertog