Embedded-AMS/EmbeddedProto #20: String length template not working for nested messages

DavidHammond123, 2021-07-20:
When a message (Message A) is created which contains a string, then the generated class has a template argument for the length of the string (this works fine).

If that message (Message A) is used in another message (Message B) as a repeated type then the new message has template arguments for how many times the original message (Message A) is repeated, and how long the string is in the original message (Message A). (This works fine).

However if the original message (A) is used in another message (message C) either just straight up (not repeated), or as part of a 'oneof' then the new message (C) WILL NOT have template arguments for the length of the string variable in the original message (A).
Attached is a sample proto file and the generated code using v2.2.
[test_proto.txt](https://github.com/Embedded-AMS/EmbeddedProto/files/6847396/test_proto.txt)
[test_h.txt](https://github.com/Embedded-AMS/EmbeddedProto/files/6847395/test_h.txt)

BartHertog, 2021-07-22:
Hello David,

Thank you for reporting this issue and providing an example! We will be looking it to this.

Best regards,

Bart

DavidHammond123, 2021-07-23:
No worries Bart, good luck with the bug search.

At the moment I have worked around it by using the 'bytes' type in my messages.

BartHertog, 2021-07-26:
This issue has been fixed. The develop branch holds the new code.

closed 2021-07-26 by BartHertog

DavidHammond123, 2021-07-28:
Thanks for fixing that bug Bart!