Embedded-AMS/EmbeddedProto #24: Oneof "selection" is not serialized when a field in the oneof is set to its default value

micemm, 2021-09-21:
When a field of a oneof block is set to its default value, the information which field in the oneof was set is lost when the message is serialized. 
(This case is explixitly mentioned in the [Documentation](https://developers.google.com/protocol-buffers/docs/proto3#oneof): "If you set a oneof field to the default value (such as setting an int32 oneof field to 0), the 'case' of that oneof field will be set, and the value will be serialized on the wire.")
This simple example demonstrates the issue:

```protobuf
message Foo {
  oneof bar{
      int32 value1 = 1;
      int32 value2 = 2;
  }
}
```
When using this message and setting one of the two value-variables to 0 (which is their default value), it is not serialized and it can not be determined which value was set on the original message after deserializing the message.
```C++
Foo testMessage;
testMessage.set_value1(0); // set a variable in the oneof to its default value

// readout which of the values is set: VALUE1
auto which = testMessage.get_which_bar();

// serialize
WriteBufferImpl writeBuffer;
testMessage.serialize(writeBuffer);
// deserialize
ReadBufferImpl readBuffer(writeBuffer.get_buffer(), writeBuffer.get_size());
Foo testMessageDeserialized;
testMessageDeserialized.deserialize(readBuffer);

// readout which value in the oneof is set -> fails: NOT_SET
auto which_deserialized = testMessageDeserialized.get_which_bar();
```
In the generated code, there is a check if the value is equal to 0 (the first condition in the if statement). If the value is inside a oneof field, this check should not be performed because when the corresponding value is selected (`which_bar_` equals `VALUE1`) the value should always be serialized, even if it is 0.
```C++
[...]
switch(which_bar_)
{
  case id::VALUE1:
    if((0 != bar_.value1_.get()) && (::EmbeddedProto::Error::NO_ERRORS == return_value))
    {
      return_value = bar_.value1_.serialize_with_id(static_cast<uint32_t>(id::VALUE1), buffer, false);
    }
    break;
[...]
```
Best regards
Michael

BartHertog, 2021-09-21:
Hello Michael,
Thank you for sending in this good description of the problem! We will fix this in a few days.
Best regards,
Bart

BartHertog, 2021-09-24:
@micemm The issue is fixed. The are available on the develop branch.

closed 2021-09-24 by BartHertog

micemm, 2021-09-28:
Thank you for the quick fix.