Embedded-AMS/EmbeddedProto #27: deserialize() does not consume payload bytes of unknown fields

jchia, 2021-11-01:
When a field id is encountered that is not defined in the .proto file, deserialize() does not consume the payload bytes that follow and instead treats the first payload byte as the start of the next field and continues processing of the fields.

deserialize() should skip the payload bytes before trying to process the next field.

BartHertog, 2021-11-01:
Hello @jchia,

Thank you for writing in this issue. I saw the previous issue coming by and I greatly appreciate you looking further and making this ticket. It is very likely we will be looking at this issue this week.

Best regards,

Bart

BartHertog, 2021-11-02:
Hello @jchia,

The issue is fixed in version 2.3.3. The code is now available. If you have any remarks let me know.

Best regards,
Bart

closed 2021-11-02 by BartHertog

jchia, 2021-11-03:
Thanks, from reading the related code, I have written #28 and #29.