Embedded-AMS/EmbeddedProto #29: ReadBufferInterface interface

jchia, 2021-11-03:
`advance()` returns void and so has no way to report an error condition, in particular, end-of-input.

Since in the recent commit a0819ae57fe440518d783c4779e52297f8e3d06e `advanced()` started to actually be used, this is when this starts to be a real issue.

The practical problem all this results in is that when skipping an known leng-delimited field, you could skip pass the end-of-input without realizing it.

BartHertog, 2021-11-03:
Hello @jchia,

Valid point. The class will be modified to return an error. 
It is a potential code breaker for current users. For this reason it will probably be released on the 3.0.0 release candidate branch.

Best regards,

Bart

BartHertog, 2021-11-04:
Hello @jchia,

In the rc-3.0.0 branch the advance() functions have been modified to include a return boolean. In the develop branch a deprecated warning has been added.

Best regards,
Bart

closed 2021-11-04 by BartHertog