Embedded-AMS/EmbeddedProto #33: Template based fixed size string/bytes becomes a bit repetitive/bugprone

davidzwa, 2022-03-13:
Before describing this feature request I want to express my gratitude for this awesome work! It's been an absolute pleasure to use this for my thesis project for making an UART CLI, LoRa communication interface and flash/SD storage interface🥇 🥇 🥇 

**Background**
From my understanding Google's Protobuf does not have the concept of fixed-size fields, which seem power this library (and with good reason!)

**Problem description**
Having multiple fixed-size fields causes the template variables to become a bit confusing. Especially when using certain message types across multiple message processor functions, I started to repeat the same template variables over and over again to eventually come up with a define to replace this with a define:
``` 
#define PROTO_LIMITS MAX_UART_FIELD1_BYTES, MAX_UART_FIELD2_BYTES, MAX_UART_FIELD3_BYTES etc etc
UartCommand<PROTO_LIMITS > uartCommand;
// Or even
#define UART_COMMAND UartCommand<PROTO_LIMITS>
```

This occurs mostly for nested fields which have nothing to do with each other. For example with OneOf I've started to notice that I had to add the same template variable multiple times - a bugprone process if you dont oversee the fields in use.
``` 
message UartCommand { 
    oneof Body { 
        RadioRxConfig rxConfig = 1; 
        RadioTxConfig txConfig = 2; 
        DeviceConfiguration deviceConfiguration = 3;
        RequestBootInfo requestBootInfo = 4; 
        ClearMeasurementsCommand clearMeasurementsCommand = 5;
        LoRaMessage transmitCommand = 6;
        LoRaReset resetRadio = 7;
    } 

    bool doNotProxyCommand = 8;
    // ...
} 
```

**Discussion**
I would like to discuss the options. Can we add or think up anything which allows us to fixate the field size **for all usages**? Maybe an optional define which gets consumed by precompilation instead of the template argument?

Im fully aware that fixated defines contradict the decoupling which is achieved by template arguments. I've found that in my application Im mostly hard-coupling the field constraints.

I guess you must have thought a lot about this by now, so Im open to hear your thoughts! 👍🏼

BartHertog, 2022-03-16:
Hello @davidzwa,

Thank you for writing in! 

It is known to us that the amount of template parameters can quickly grow and make defining objects in some cases obscure. We find it a disadvantage of the fixed length arrays as well but outweighed it against the advantage of static allocation. However we have been considering two possible improvements:
1. Working more with interface classes in the messages so that the functions in a message are less depended on the template parameters. This makes working with a message a bit easier although the definition still requires the parameters. 
2. Using Google protobuf custom options to define the length of a repeated, string or bytes field in the .proto file. This feature has been implemented and is available in the RC3.0.0 branch.  

The plan was to release RC3.0.0 about now but this has been delayed by a few weeks. The proper documentation is still being written but we have the following notes:

You have to do two things:
1. After updating the latest commit of RC3.0.0, run the setup script in the EmbeddedProto root folder. This will generate a python file from generator\embedded_proto_options.proto. On windows machines please provide the correct folder parameter, see README.
2. Include embedded_proto_options.proto in all the files using the maxLength parameter like:

```protobuf
syntax = "proto3";
// This is required for the Embedded Proto protoc plugin to understand the option.
import "embedded_proto_options.proto"; 

message Dummy{
  repeated uint32 a = 1 [(EmbeddedProto.options).maxLength = 10];
  repeated uint32 b = 2;
}
```
This will fixate the length of field a. You will not get a template parameter for this.

If you would like more examples you can look at the unit tests related to [test/proto/field_options.proto
](https://github.com/Embedded-AMS/EmbeddedProto/blob/rc-3.0.0/test/proto/field_options.proto).

Please note that the RC3.0.0 branch requires at least C++17.

Any other ideas for improvements are welcome.

Best regards,

Bart

davidzwa, 2022-03-16:
Hi Bart,

Awesome and really cool that you're already working on this. Feel absolutely free to close this issue as you have proposed a decent solution.

About C++17, I just checked and it seems we're already on it! All good and thanks for the heads up.
> set_property(TARGET ${PROJECT_NAME} PROPERTY CXX_STANDARD 17)

Kind regards,
David

BartHertog, 2022-03-25:
Hello David,

Your welcome. I will leave this ticket open for others to read the solution. This until the new version with documentation has been released.

Best regards,

Bart

BartHertog, 2022-04-28:
V3 with documentation is out now and available on Github and the website.

davidzwa, 2022-04-28:
Congrats!!