Embedded-AMS/EmbeddedProto #34: Cannot have a 'repeated' enum field

Imroy, 2022-03-31:
The RepeatedField template class requires its DATA_TYPE field to be a type derived from the Field class. This rules out having a repeated enum field in a message.

e.g
```
enum QoS {
  none = 0;
  good = 1;
  better = 2;
}

message Subscribe {
  repeated string topics = 1
  repeated QoS qoses = 2;
}
```

Results in:
```
../../EmbeddedProto/RepeatedField.h:52:71: error: static assertion failed: A Field can only be used as template paramter.
```

BartHertog, 2022-03-31:
Hello @Imroy,

Thank you for writing in. We will be looking into this issue shortly.

Best regards,

Bart

BartHertog, 2022-03-31:
Hello @Imroy,

The issue was confirmed on the _develop_ branch. The release candidate branch rc-3.0.0 does not give the error. Could you try using the rc-3.0.0 branch up until we release the new major version?

Using an enum as a repeated field was not included in the unit tests. We will add these tests on the rc.3.0.0 branch in the coming week. 

Best regards,

Bart

BartHertog, 2022-04-05:
Hello @Imroy,

The unit tests have unearth some issues also present branch rc-3.0.0. We are working on this.

Best regards,
Bart

Imroy, 2022-04-07:
The rc-3.0.0 branch doesn't install for me. When running ```setup.sh``` I get:
```
ERROR: Could not find a version that satisfies the requirement pkg_resources==0.0.0 (from versions: none)
ERROR: No matching distribution found for pkg_resources==0.0.0
```

Removing the ```pkg_resources=0.0.0``` line from ```requirements.txt``` fixed that.

After that, the new ```IS_BASIC_TYPE``` work in ```RepeatedField``` doesn't work.
```
.../EmbeddedProto/RepeatedField.h:67:72: error: static assertion failed: A Field can only be used as template paramter.
     static_assert(std::is_base_of_v<::EmbeddedProto::Field, DATA_TYPE> || IS_BASIC_TYPE,
```

BartHertog, 2022-04-07:
Hello @Imroy,

Yes the IS_BASIC_TYPE was a problem together with enums not using the FieldTemplate class. They do now in our private issue branch. Also custom trait has been written yesterday and we will review this code today.

The installation problem is new. This will be checked in a new (internal) ticket. 

Best regards,

Bart

BartHertog, 2022-04-07:
Hello @Imroy,

The changes have been pushed to rc-3.0.0. We are considering fixing this on 2.x.x but this requires a different solution. In this part of the code v2 and v3 are different. We will keep you posted.

Best regards,

Bart

BartHertog, 2022-04-08:
The modifications have been made on the develop branch and released to 2.4.0.

closed 2022-04-08 by BartHertog