Embedded-AMS/EmbeddedProto #38: plugin is not working

xakepp35, 2022-05-14:
Tried to review, if it can be used for cpp project. Running it caused error:
```
--eams_out: Embedded Proto error - list.remove(x): x not in list
```

xakepp35, 2022-05-14:
looks like I can correct error message a bit - "Embedded Proto error - python is not in the deps list of a serious cpp project, especially when you set pricing plans for commercial usage!"

BartHertog, 2022-05-14:
Hello @xakepp35,

Thank you for writing in! You effort is appreciated. 

What likely caused this problem was including an empty *.proto file. The error message did not represent that at all.
This issue was already resolved by checking the length of the list and an unit test was added for the case of an empty *.proto file. 

These changes are available on the development and the v2-hotfixes branch. However they were not merged into master as a formal release. This will be done in the coming days.

If you have any questions, let me know.

Best regards,

Bart

BartHertog, 2022-05-17:
Hello @xakepp35,

Release 3.0.1 features the solution to this issue.

Best regards,

Bart

closed 2022-05-17 by BartHertog