Embedded-AMS/EmbeddedProto #39: Support repeated enum fields

ChiSaw, 2022-05-31:
Hello,

We noticed that repeated enum fields are currently not supported

```proto
syntax = "proto3";
package test;
import "embedded_proto/options.proto";

enum Config {
  CONFIG_UNSPECIFIED = 0;
  CONFIG_UP = 1;
  CONFIG_DOWN = 2;
}

message PayloadEnvelope {
  repeated Config config = 1;
}
```

crashes with

```log
In file included from ./../../src/anchor/../../third-party/embeddedproto/src/RepeatedFieldFixedSize.h:34,
from ./../../src/anchor/../../test.h:30,
./RepeatedField.h: In instantiation of 'class EmbeddedProto::RepeatedField<test::Config>':
./RepeatedFieldFixedSize.h:52:9: required from 'class EmbeddedProto::RepeatedFieldFixedSize<test::Config, 4>'
required from here
./RepeatedField.h:93:74: error: static assertion failed: A Field can only be used as template paramter.
93 | static_assert(std::is_base_of_v<::EmbeddedProto::Field, DATA_TYPE> || IS_BASIC_TYPE,
| ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
```

The only workaround is 

```proto
syntax = "proto3";
package test;
import "embedded_proto/options.proto";

message Config
  enum ConfigType {
    CONFIG_UNSPECIFIED = 0;
    CONFIG_UP = 1;
    CONFIG_DOWN = 2;
  }
  ConfigType config = 1;
}

message PayloadEnvelope {
  repeated Config config = 1;
}
```

which costs 1-2 bytes for each repeated element additionally. Hence, direct support of repeated enum fields would save quite a few bytes in the serialized byte stream

Best,
ChiSaw

BartHertog, 2022-05-31:
Hello @ChiSaw,

Thank you for reporting this issue. We will be looking into it shortly. I suspect there is something missing for the max length definition. The case without maxLenght is covered by an unit test but this one is missing.

Best regards,

Bart

ChiSaw, 2022-06-08:
Hi @BartHertog,

indeed we had an old version back from February, it is working as expected! Thanks!