Embedded-AMS/EmbeddedProto #40: Windows installation instructions not working

as08gf9uw0tjawtegi, 2022-06-14:
When following the installation instructions I get the following output in Powershell: 

`EmbeddedProto> .\setup.bat C:\ProgramData\chocolatey\lib\protoc\tools\include\google\protobuf
Collecting Jinja2==3.0.3
  Using cached Jinja2-3.0.3-py3-none-any.whl (133 kB)
Collecting MarkupSafe==2.0.1
  Using cached MarkupSafe-2.0.1-cp310-cp310-win_amd64.whl (15 kB)
**ERROR: Could not find a version that satisfies the requirement pkg_resources==0.0.0 (from versions: none)**
**ERROR: No matching distribution found for pkg_resources==0.0.0**
**'.\venv\Scripts\deactivate.bat)' is not recognized as an internal or external command,
operable program or batch file.**`

BartHertog, 2022-06-14:
I can reproduce it. I will give you an update when I found a solution.

BartHertog, 2022-06-14:
The error about the `)` is prevented by adding a space between `.bat )`. However a more serious problem is that the virtual environment is not activated and the packages are installed globally. 

As the second work around I suggest to you to day you can execute the following commands by hand. In the EmbeddedProto folder open a Powershell terminal and execute:
```
.\venv\Scripts\Activate.ps1
pip install -r requirements.txt
deactivate
```
After this you can rerun the setup.bat file again. It will still complain but generating code with Embedded Proto will work.

I will keep the issue up to date on changes related.

as08gf9uw0tjawtegi, 2022-06-14:
I ran into another issue (still Windows) where the path specified in `--eams_out=` when running protoc in Powershell can't handle any path that has spaces. I tried writing the path as "$pwd\generated\_src",  but it seems to just ignore the `"` characters (which usually in Windows would make a path with spaces work). 

I tried to rename the directory so that it wouldn't have spaces and tried to compile again using protoc (after rerunning all of the suggested steps above) and it seems it can't find the options file. I get the following output: 
`EmbeddedProto> protoc --plugin=protoc-gen-eams=protoc-gen-eams.bat -I C:\ProgramData\chocolatey\lib\protoc\tools\include\google\protobuf -I "$pwd\generator\" -I "$pwd" --eams_out="$pwd\generated\_src" serialize_deserliaze.proto                                --eams_out: Embedded Proto warning - Unable to load the Embedded Proto Options file. Did you run the setup script? The script should generate the file. This run we will not include the options specified in your *.proto file. `

BartHertog, 2022-06-15:
Will take it into account!
Just to check, does your setup work when renaming the folder to one without spaces?

as08gf9uw0tjawtegi, 2022-06-15:
The setup runs fine for a path with or without spaces: except the Embedded proto options file somehow doesn't get picked up correctly (regardless of whether the path has spaces or not). 

It's just the protoc command's --eams_out= options that doesn't seem to handle a path with spaces correctly.

as08gf9uw0tjawtegi, 2022-06-15:
Indeed the problem was importing the embedded_proto_options_pb2 module (line 115 in protoc-gen-eams.py). 

The reason that this import failed: 
`embedded_proto_options_pb2.py", line 5, in <module> from google.protobuf.internal import builder as _builder                                                            ImportError: cannot import name 'builder' from 'google.protobuf.internal' `

Which in turn seems to be caused by the installed python protobuf module version being too low (under version 3.2). I managed to get the whole thing working by `pip install --upgrade protobuf ` (version 4.21.1 which is the newest at the time of writing). 

TLDR/Rootcause: in requirements.txt the `protobuf==3.19.4` line should have a higher version number (probably higher than 3.2 I'm guessing).

as08gf9uw0tjawtegi, 2022-06-15:
> With regards to installing a different version of the protobuf module. I do not fully understand your comment as version 3.2.x (3.02.x) is below 3.19.x. I have protoc-21.1-win64 running here and the python protobuf module 3.19.4 and can not reproduce your error.

I'm confused by the different version numbers, all I know is it worked after I updated the protobuf python package. 

> What do you mean by directories included in the git clone? The user is able to output their generated code anywhere they would like. It is not contained to the EmbeddedProto repo. Actually I would suggest not outputting your code in the EmbeddedProto repo folder but in a separated build/generated folder.

Oh, I wasn't very clear here (and confused myself). Let me retry: it would be nice if the suggested usage command for protoc in the EmbeddedProto git README.md (https://github.com/Embedded-AMS/EmbeddedProto) would work out of the box without complaining about a dir not existing. Meaning that the directory used in the example should exist: `--eams_out=.\generated_src`

BartHertog, 2022-06-16:
If you want I can explain the difference but I feel this thread is not the location.

I have no solution for the white spaces yet.  I have more time to look at this early next week. In advance thank you for your patience.

BartHertog, 2022-06-17:
Now on the develop branch you will find the following issues resolved:
1. pkg_resources==0.0.0 has been removed from the requirements.txt file.
2. '`.\venv\Scripts\deactivate.bat)` is not recognized as an internal or external command, operable program or batch file.' The python packages are now properly installed in the virtual environment and not globally.
3. With regards to the version mismatch between Protoc and the python protobuf module. There is now a version check in our plugin code. Please note Embedded Proto now only works with protoc 3.19.4. As of 3.20.0 the google protobuf python module was rewritten by the google team, introducing breaking changes. This is why you got the import error `embedded_proto_options_pb2.py", line 5, in <module> from google.protobuf.internal import builder as _builder ImportError: cannot import name 'builder' from 'google.protobuf.internal'`. The older protobuf module was unable to deal with the new protoc compiler. Eventually we will update to the latest greatest but this needs to be tested properly with these external changes to our dependency. Please use protoc version 3.19.4

With regards to the spaces in folder paths. This is not within my direct influence. This is an issue in protoc, before our plugin Embedded Proto plugin is even called.

closed 2022-06-17 by BartHertog