Embedded-AMS/EmbeddedProto #41: Support for DebugString()

sdwhitney, 2022-06-23:
I noticed that the header file for the embedded output does not provide DebugString() to provide a human-readable form for debugging purposes.  Do you have any recommendations on how this could be provided, whether automatically generated by your protoc plugin, or manually on an as-needed basis?  Thanks!

BartHertog, 2022-06-23:
Hello @sdwhitney,

Thank you for writing in!

A function like DebugString() would require a function like sprintf(). These types of functions are quite big, they take up a lot of flash. For this reason it is not in our code at this moment. I will put a ticket on the backlog to evaluate if this can be done in an efficient manor. 

The way I debug now is with a desktop application which interprets the debug output of the mcu or listens in on the data bus. This external program can be written in any programming language supported by protobuf. The google code almost always has a function which converts the data to human readable format like JSON.

I hope this answers your question. If you have an more, let me know.

Best regards,

Bart

sdwhitney, 2022-06-23:
Hi Bart,

I understand your reasoning. Printf() can definitely get big!

Your idea of performing this on the desktop side makes sense. Our
microcontroller will be communicating with a Linux system-on-module over a
UART.

I know there are reduced versions of printf() for embedded systems. Some
remove printing of floats, doubles, and pointers to save space. Maybe
having a way to control whether DebugPrintf() gets generated in the
embedded output would be useful, perhaps controlled by a #define USE_PRINTF
to let the user decide if they want to enable DebugString() on the embedded
side?

Thanks for the quick response too!

Best regards

Scott

BartHertog, 2022-06-24:
Hello Scott,

Yes a customer of Embedded AMS is using https://github.com/eyalroz/printf as an alternative printf. This is a reasonable alternative but I would prefer the user of Embedded Proto making that decision.  

Considering how to enable the user some flexibility I am considering adding a set of constexpr which hold the name of each variable as a static C style string. Possibly link the name with the field id number and some way to get the correct value. This is last point is not clear yet. Adding some function pointer will introduce complexity and will take up additional ram. This is the point I am still considering on how to approach.

Best regards,

Bart

BartHertog, 2022-07-12:
Hello Scott,

With regards to this ticket I have done some coding today. For each variable in a message a static character array is now defined with the name of the variable. There is also a function named `field_number_to_name(FieldNumber)` in each of the message objects. This function returns the variable name of a field based on the FieldNumber enum value.

I hope this helps you.

Best regards,

Bart

sdwhitney, 2022-07-13:
Hi Bart,

Thank you for giving this some attention!

I'm a little confused, though. Can you give me an example of how I could
use this feature to create a DebugString() functionality?

Kind regards and sincere thanks for your help!

Scott

sdwhitney, 2022-07-15:
No problem, Bart!  I appreciate that you are working toward something to satisfy my request, and think that it might end up being a very desirable improvement to all users of Embedded Proto!

Best regards,

Scott

BartHertog, 2022-10-25:
Hello @sdwhitney,

Good news! I have implemented a decent to_string() function! It is now available on the develop branch. 
An example of what a message could look like is:
```
{
  "u": 1.000000,
  "nested_a": {
    "x": [
           1,
           2,
           3
         ],
    "y": 1.000000,
    "z": 1
  },
  "v": 1
}
```

Take into account the following:
- You need to use the `string_view` struct defined in the Defines.h header.
- Use the function `string_view to_string(::EmbeddedProto::string_view& str) const` to print the message to str. The data left in the string array is returned.
- The function ignores anything related to default values. Basic fields are displayed even when not set.
- Fields in a oneof are not displayed when set.
- This requires C++17 and up.
- Use the build flag MSG_TO_STRING to enable this functionality. It uses snprintf so it is not advised to use this for production.