Embedded-AMS/EmbeddedProto #42: [(EmbeddedProto.options).maxLength = xxx] causes weird memory scaling issue?

as08gf9uw0tjawtegi, 2022-07-06:
I ran into a weird issue and I'm not sure how to check whether it's an issue with my SoC (hardware) I'm using or an issue with EmbeddedProto. 

To replicate the following could be tried: 

Create a new message with:

`message Test {
   repeated bytes testbytes = 1 [(EmbeddedProto.options).maxLength = 5]; 
}`

Then in C++ do a sizeof(Options::Test); to check the size of the resulting EmbeddedProto Test class. You'll get a reasonable size. 

Next try: 

`message Test {
   repeated bytes testbytes = 1 [(EmbeddedProto.options).maxLength = 512]; 
}`

In my case the sizeof() method returned a size of 266368 bytes, which is way more than I would expect and I'm not sure what's causing it. It also caused my code to crash at a seemingly completely random point (way before the EmbeddedProto code would get called) which smells of memory leak or just the flash/RAM being too full (in my case I only have 64kb RAM).

BartHertog, 2022-07-07:
Will be looking into this. I will get back to you when I have some news.

BartHertog, 2022-07-07:
Actually 266368 is about the correct size. What happens here is that both `bytes` and the `repeated` part of the definition are arrays. And both use the maxLength value. So in essence `repeated bytes` defines an array of an array both of size 512 resulting in 512 * 512 bytes being allocated with some overhead for each bytes class.

If you need multiple bytes you can drop the `repeated` from the definition. 

If it is your goal to do have an array of byte arrays I would suggest eigther:  not using maxLength in the protofile but using the template option