Embedded-AMS/EmbeddedProto #46: Installation not working on Ubuntu - Is my protobuf installed wrong?

burgerga123, 2022-09-12:
I'm getting the following error when I try to set things up on Ubuntu:
```
> python setup.py

================================================================================
|                                                                              |
| Three simple things you can do to help improve Embedded Proto:               |
|  * Give private feedback:                                                    |
|        https://EmbeddedProto.com/feedback                                    |
|  * Report an issue in public on Github:                                      |
|        https://github.com/Embedded-AMS/EmbeddedProto/issues                  |
|  * Stay up to date on Embedded Proto via our User mailing list:              |
|        https://EmbeddedProto.com/signup                                      |
|                                                                              |
================================================================================

Checking the version of Python
Checking your Protoc version.
Creating a virtual environment for Embedded Proto.
Installing requirement Python packages in the virtual environment.
Build the protobuf extension file used to include Embedded Proto custom options.
Error: Command '['protoc', '-I', 'generator', '--python_out=generator', 'embedded_proto_options.proto']' returned non-zero exit status 1.
Setup completed with success!
```

I installed protobuf following the instructions here: https://github.com/protocolbuffers/protobuf/blob/main/src/README.md
As a result protoc is installed `/usr/local/bin` and I can see all the include files in `/usr/include/google/protobuf`

I also tried `python setup.py --include /usr/include/google/protobuf` with a similar error:
`Error: Command '['protoc', '-I', 'generator', '--python_out=generator', 'embedded_proto_options.proto', '-I', '/usr/include/google/protobuf']' returned non-zero exit status 1.`

The install instructions for EmbeddedProto seem to assume that protobuf will be installed in the home directory and there will be a folder called "include". But I'm not sure how to install protobuf in this way... What am I doing wrong?

BartHertog, 2022-09-12:
Hello @burgerga123,

Thank you for writing in!

I noted a few things I could improve. The actual error you have is not reported, and it states that the setup was completed with success, which it obviously did not. Could you try to get the actual message for me?

You enter the Embedded Proto folder and run:
```
protoc -I generator --python_out=generator embedded_proto_options.proto
```
Let me know what the output is.

If this worked correctly there should be a new file: `generator/embedded_proto_options_pb2.py`. 

I will work on the script to report the actual error in the future.

Best regards,

Bart

burgerga123, 2022-09-12:
Hi Bart, here is the result:

```
> protoc -I generator --python_out=generator embedded_proto_options.proto
google/protobuf/descriptor.proto: File not found.
embedded_proto_options.proto:3:1: Import "google/protobuf/descriptor.proto" was not found or had errors.
embedded_proto_options.proto:11:8: "google.protobuf.FieldOptions" is not defined.
```

It seems that the file it can't find is on my system here:
```
> cd /usr/include/google/protobuf  
> ls
any.h                  extension_set.h                   map_field_lite.h      struct.pb.h
any.pb.h               extension_set_inl.h               map.h                 struct.proto
any.proto              field_mask.pb.h                   map_type_handler.h    stubs
api.pb.h               field_mask.proto                  message.h             text_format.h
api.proto              generated_enum_reflection.h       message_lite.h        timestamp.pb.h
arena.h                generated_enum_util.h             metadata.h            timestamp.proto
arena_impl.h           generated_message_reflection.h    metadata_lite.h       type.pb.h
arenastring.h          generated_message_table_driven.h  parse_context.h       type.proto
descriptor_database.h  generated_message_util.h          port_def.inc          unknown_field_set.h
descriptor.h           has_bits.h                        port.h                util
descriptor.pb.h        implicit_weak_message.h           port_undef.inc        wire_format.h
descriptor.proto       inlined_string_field.h            reflection.h          wire_format_lite.h
duration.pb.h          io                                reflection_ops.h      wrappers.pb.h
duration.proto         map_entry.h                       repeated_field.h      wrappers.proto
dynamic_message.h      map_entry_lite.h                  service.h
empty.pb.h             map_field.h                       source_context.pb.h
empty.proto            map_field_inl.h                   source_context.proto
```

BartHertog, 2022-09-12:
Hi burgerga123,

If you look at `embedded_proto_options.proto` this file includes `google/protobuf/descriptor.proto`. Possibly you just have to supply the path to the include folder: `python setup.py --include /usr/include/` without the `google/protobuf`.

Best regards,

Bart

P.S. As a reference on my system, the include folder is located in `/usr/local/include/google/protobuf.

burgerga123, 2022-09-13:
Hey Bart,

I got it working, but it took a few more steps.

The command you suggested allowed the setup to complete correctly.
I moved the `protoc-gen-eams` file to `/usr/local/bin` so that it's in my PATH, but then I got an error that it's not found.
So I ran the command with manually specifying: `--plugin=/usr/local/bin/protoc-gen-eams`
This seemed to get me to the next error, that `/usr/local/bin/venv/bin/python` could not be found. I see the venv folder in the EmbededProto git so I moved that to `/usr/local/bin` as well.
Then had the same issue with `/usr/local/bin/generator/protoc-gen-eams.py`. So I moved the `generator` folder to bin.
Last issue, it couldn't find the folder `generated_src`. Apparently this needs to be made ahead of time and it won't make the directory if it doesn't exist?

With all these fixes I can now build my proto header file!

Can I suggest adding these steps to your readme?

- Better clarity on the protobuf include folder. The protobuf repository doesn't have a folder called `protobuf/protoc-21.5/include` and the default protobuf installation puts it in `/usr/include`
- Need to move `protoc-gen-eams`, `venv` folder, and `generator` folder to `/usr/local/bin`
- Manually specify the path to `protoc-gen-eams` in the protoc command since it doesn't seem to use the path in this command

closed 2022-09-13 by burgerga123

BartHertog, 2022-09-13:
Hello @burgerga123,

I am glad to hear that you got it working! 

I now understand you were trying to install Embedded Proto globally. Embedded Proto was designed with a different use case in mind. It is designed to be used as a submodule to your project and be contained in that local folder. I now understand that this is what should be better explained in the documentation and readme. 

To give you a perspective, I work with many different projects, all with different versions of protobuf and thus also Embedded Proto. Also, I have to maintain older versions of released code. By using Embedded Proto as a submodule I can switch branches of the main project and update submodules to match that (older) version.

This is also where the Python virtual environment comes into play. This is a local representation of a given python version and packages installed in that environment. By running the setup script matching packages are installed. Placing that virtual environment in your path could cause problems with your python installation and future updates. Please consider removing it from your global path. [Here](https://docs.python.org/3/tutorial/venv.html) you can find more information about python virtual environments.

Again the local installation is what I assumed when giving the example `protobuf/protoc-21.5/include`. It would have been better if I specified `~/protobuf/protoc-21.5/include`. I will add this in the readme. In this case, the folder '~/protobuf' is a folder with multiple versions of protoc you compiled from source. 
```
~/protobuf/protoc-3.20.1
~/protobuf/protoc-21.0
~/protobuf/protoc-21.5
```

Hope this helps you understand the use case of embedded proto. You could look at one of the examples of Embedded Proto. They all have the submodule setup.

Best regards,

Bart