Embedded-AMS/EmbeddedProto #49: FieldString operator= triggers stringop-truncation error

wuyuanyi135, 2022-10-01:
https://github.com/Embedded-AMS/EmbeddedProto/blob/3fe792288b289f24c2789ece31e36dd25668d8cb/src/FieldStringBytes.h#L300

As `strlen` does not count the tailing `\0`, the compiler check will complain that this `strncpy` may be truncated.

Changing this line to `strncpy(this->get(), rhs, this->get_length() + 1);` will suppress this warning/error.

BartHertog, 2022-10-03:
Hello @wuyuanyi135,

Thank you for writing in! 
Looks like this is indeed a problem. I do not see any check if the string actually fits in the destination so I will include that as well.

Best regards,

Bart

BartHertog, 2022-10-04:
The changes are in the develop branch. I had to check if the null character fits in the destination so it is not an exact copy of your suggestion.

Best regards,

Bart

closed 2022-10-04 by BartHertog