Embedded-AMS/EmbeddedProto #50: oneof containing a message causes `--eams_out: Embedded Proto ERROR`

wuyuanyi135, 2022-10-10:
When compiling the following `.proto`
```protobuf
syntax = "proto3";
import "embedded_proto_options.proto";

message Periodic {
}

message BuzzerCommand {
  message Request {
    oneof buzzer_control {
      Periodic blink = 1;
    }
  }
}
```

with the command
```
protoc --plugin=protoc-gen-eams.bat --eams_out=. local_command.proto
```

The following error is thrown 
```
--eams_out: Embedded Proto ERROR - Messages with repeated, string or byte fields use template parameters to define their length.For some reason it was not to add all required template parameters.
```

If oneof contains a message this error is thrown. When compiling this `proto` for other languages no error occured.

BartHertog, 2022-10-10:
Hello @wuyuanyi135,

Thank you for writing in.

This error is misleading or not complete. What is happening is that defining nested messages has only minimal support. If you move the definition of the Request message out of BuzzerCommand the error will be gone.

I will look into detecting this case and reporting it with an appropriate error message.

Best regards,

Bart

wuyuanyi135, 2022-10-10:
@BartHertog Thanks for your explanation. Does the documentation mention the limitation of the nested message definition? Is `oneof` the only place where nested message should be prevented?

wuyuanyi135, 2022-10-10:
@BartHertog Hello. I found that the behavior is not caused by the nested message.

Try compiling the following proto and change the commented line.
* If the message's name is `BCommand`, `BuzzCommand`, BuzzerCommand`, etc, the compile will fail
* If the message's name is `Command`, `InfoCommand`, etc. it compile successfully.

I think it is some inconsistency in the compiler.

```
syntax = "proto3";
import "embedded_proto_options.proto";

message Capability {
}

message BCommand { // <<------- This line
  message Request {}
  optional Request request = 1;

  message Response {
    string model = 1 [(EmbeddedProto.options).maxLength = 20];
    string version = 2 [(EmbeddedProto.options).maxLength = 10];
    Capability cap = 3;
  }
  optional Response response = 2;
}

```

wuyuanyi135, 2022-10-10:
Ok. The problem is more strange than I throught: 
If there are nested messages, and the nested message is another message, and if the first letter of the wrapping message name is `A-C`, it will fail to compile. For example, 'ACommand`, `BCommand`, and  `CCommand`,  will fail but 'DCommand' can pass the compilation. 
```
syntax = "proto3";
import "embedded_proto_options.proto";

message Capability {
}

message DCommand {
  message nested {
    Capability c = 1;
  }
}

```

BartHertog, 2022-10-10:
O wow, 

Top discovering this querk! I will debug it tomorrow. 

Best regards,

Bart

wuyuanyi135, 2022-10-10:
I gave it a quick debug. It seems like the order of `register_template_parameters` will affect the result.

wuyuanyi135, 2022-10-10:
https://github.com/Embedded-AMS/EmbeddedProto/blob/3fe792288b289f24c2789ece31e36dd25668d8cb/generator/protoc-gen-eams.py#L113
This line looks like the problem. If I understand correctly, you want to run the root registration multiple (three) times to ensure some late registration of types will be done in the later retries. 
However, the logic here with boolean will always be False if the previous trial did not register all types. I removed the `and all_parameters_registered` at the end of the line, the problem is gone for me. I am not sure if this will break some other parts of the code.

wuyuanyi135, 2022-10-10:
> When compiling the following `.proto`
> 
> ```proto
> syntax = "proto3";
> import "embedded_proto_options.proto";
> 
> message Periodic {
> }
> 
> message BuzzerCommand {
>   message Request {
>     oneof buzzer_control {
>       Periodic blink = 1;
>     }
>   }
> }
> ```
> 
> with the command
> 
> ```
> protoc --plugin=protoc-gen-eams.bat --eams_out=. local_command.proto
> ```
> 
> The following error is thrown
> 
> ```
> --eams_out: Embedded Proto ERROR - Messages with repeated, string or byte fields use template parameters to define their length.For some reason it was not to add all required template parameters.
> ```
> 
> If oneof contains a message this error is thrown. When compiling this `proto` for other languages no error occured.

By the way, the `oneof` of nest message seems to be supported after removing the line mentioned above.

BartHertog, 2022-10-11:
Hello @wuyuanyi135,

After debugging a bit, I found the problem. The problem was with the topo sorting algorithm. This determines the order of the message definitions in a header file. For example, if message A depends on message B, message B should be defined first in the header. 

In the case you presented, `BuzzerCommand::Request` depends on `Periodic`. This means `BuzzerCommand` indirectly depends on `Periodic`. Passing down this dependency from `BuzzerCommand::Request` to `BuzzerCommand` was not done. 

The result was that the definition of `BuzzerCommand` was placed first in the header file before `Periodic`. 

You got the actual proper error message as it could not resolve the usage of `Periodic` in `BuzzerCommand::Request` as `Periodic` was not yet defined. This is what the boolean indicates you removed.

Concerning renaming the message to get it to work. For some reason, the order changed when changing the name. Maybe the internal workings of the toposort algorithm does something based on the name.

The fix and unit tests are now available on the develop branch.

Best regards,

Bart

closed 2022-10-11 by wuyuanyi135