Embedded-AMS/EmbeddedProto #52: Dynamically allocate memory for message string and bytes fields

wuyuanyi135, 2022-10-10:
Currently, this library adopts the memory model of statically sized `std::array` for size-variable fields like string and bytes. This benefits most embedded development workflow since no heap allocation occurs, hence the message creation is a break-or-make. 

However, I am currently designing some systems involving extreme message sizes. For example, a `bytes` field can be as short as 1 byte of payload or as long as 100k bytes. Under current memory model, I have to create the message with the max possible size: 100k. Considering that I may also have a queue for the messages, it can easily bloat the memory usage to MBs, even if I only have a few packet with the shortest bytes field. 

My suggestion is to provide a protobuf option or an optional template argument, to allow user to specify the desired internal memory container. For example, std::vector. 

Introducing the dynamically allocated fields comes with the cost of potentially failed memory allocation.  During encode and decode the error can be captured using the return value. However, during the message creation (assigning, modifying the message content), the possible memory allocation may fail. To capture that, the standard way is to use try-catch exception, which seems also should be avoided for embedded environment due to its overhead.

This feature may help extending the application on the platform with moderate RAM space while involving more complex communication and data exchange. For example, my ESP32-S3-WROOM-1-N8R8 has hundreds KB of internal RAM plus 8 MB of external RAM. The operating system infrastructure prevents me to use the Google implementation of protobuf. Therefore, I am doing some research to see if this library will suite my application.

What's your opinion regarding the dynamically-sized string and bytes fields?

BartHertog, 2022-10-11:
I have no experience yet with the ESP32 suite and did not know it has this much memory. This use case had thus not crossed my mind. I personally would advise against using dynamic memory allocation in Embedded Proto and in general as many coding standards do not allow for this.

However, there are alternatives that I suspect give the behavior you desire. In the[ other ticket ](https://github.com/Embedded-AMS/EmbeddedProto/issues/51#issuecomment-1274599147) I already mentioned using std::string_view. 

For repeated fields, I am thinking of using something similar or maybe using iterators. 

This all should also apply to repeated fields.

wuyuanyi135, 2022-10-11:
@BartHertog Thanks for your response. I could confirm that this library works flawlessly on esp-idf framework (presumably should also work on Arduino for ESP). You could consider including the ESP platform in your support list. 

The ESP series has a rich list of communication capabilities (WiFi, Bluetooth, Ethernet, USB, ...) And this library serves well as a application layer to implement  a consistent RPC between devices.

BartHertog, 2022-10-12:
I am thrilled to hear it works on ESP. It is in the backlog to make a demo with Embedded Proto on one of the Espressif chips, but I did not get around to doing that.

wuyuanyi135, 2022-10-12:
Yes. I'm glad that this library has been implemented in a platform agnostic manner. I will try to implement the dynamic allocation feature in my form these days. In the meantime if you want some help to make the demo project of ESP series, I would like to get involved.

BartHertog, 2022-10-12:
For another ticket, I started working on a simple implementation of string_view. This class is only available as of C++17 and currently Embedded Proto is compatible with C++11. If you want I can share the definition.

A little help with a simple ESP32 example would be great! Do you have an example in mind?

wuyuanyi135, 2022-10-13:
@BartHertog By referring to the `string_view`, do you mean not to store the actual data inside a decoded message, but using a `string_view` to point to the data block stored in the `ReaderBufferInterface`? This is indeed a very smart way to prevent duplicated memory in both decoder and message. But this will also require the control of lifetime of the reader interface, otherwise the `string_view` will be pointing an invalid memory region, correct?

closed 2022-10-15 by wuyuanyi135