Embedded-AMS/EmbeddedProto #53: out-of-bound get_const on FieldBytes should raise error rather than return 0

wuyuanyi135, 2022-10-15:
https://github.com/Embedded-AMS/EmbeddedProto/blob/3fe792288b289f24c2789ece31e36dd25668d8cb/test/test_string_bytes.cpp#L281

If `msg` was cleared, the length is set to zero, then accessing `msg.get_const(0)` should be considered invalid. This library seems not  using C++ exception but I think this situation should be asserted rather than defined (to return 0) to prevent potential bugs.

What's your opinion?

BartHertog, 2022-10-17:
Hello @wuyuanyi135,

It is correct no exceptions or asserts are used. Most microcontrollers have no support for these. When they hit one of these they will go into a hard fault. This will yield undefined behavior with possibly dangerous situations. Think of controlling a motor, while it is running this occurs. Will the motor keep running, will it stop, are safety stops still operational?

To prevent such situations it was chosen to return the value nearest to the index or zero.

A safer function would be something like: `error get_const(const int32_t index, const TYPE& value);` In this case an error could be returned.

BartHertog, 2022-10-25:
The function mentioned above is now implemented for String, Bytes, and repeated fields. You can find this code on the develop branch.