Embedded-AMS/EmbeddedProto #54: Can't install EmbeddeProto

Stipa88, 2023-03-07:
I use windows 10 Pro, 64bit. 
I use ESP_IDF for ESp32, so Python, Git...I have it installed from before.
By opening CMD, and the folder "cd EmbeddedProto", calling command "python setup.py" or "python setup.py --include ~/protobuf/protoc-21.5/include" it generates me an error: (see Picture)
[image]
Please help.

BartHertog, 2023-03-08:
Hello @Stipa88,

Thank you for writing in. Sorry to hear you are having problems installing it. Let's figure this out!
From what I see now, I am thinking you might be using a newer version of python with a change in it. Which version are you using?

Best regards,

Bart

BartHertog, 2023-03-08:
Looking at the docs venv still has the create function. Is Pip installed on your system?