Embedded-AMS/EmbeddedProto #59: Setup Script Include Not Overriding System Protoc

ttchalakov, 2023-05-28:
I am developing on Ubuntu 20 x64 and am following the installation instructions. My system already has another version of protoc installed that is used for another dependency so I want to use the `--include` option in the setup.py script to specify the correct version of protoc (v21.5) but the setup script is not using the new version. I am installing protoc by unzipping this download of [protoc-21.5-linux-x86_64.zip](https://github.com/protocolbuffers/protobuf/releases/v21.5). When I unzip this file, it produces an include and bin folder. I then provide the path in the setup.py script with the include option to the include folder but the script still sees the version that was already installed on my system before. I am not sure if this is because I am installing protoc incorrectly or if the script is not able to see the executable for protoc v21.5 because the include folder doesn't contain the executable for protoc, the bin folder contains it.

adisbladis, 2023-06-01:
You most likely just need to ensure your desired `protoc` implementation shows up first in `$PATH`.

BartHertog, 2023-07-06:
Hello @ttchalakov,

Thank you for writing in. From the top of my head, `--include` only specifies the directory of the .proto file required for building. Like @adisbladis suggested, the protoc binary is taken from the path. Have you been able to add your new install to the path and make it work like that?

Best regards,

Bart

ttchalakov, 2023-10-20:
Sorry for the delayed reply. I solved my problem by uninstalling through `apt` the system's `protobuf-compiler`, compiled from source the version I wanted to use, and then ran `setup.py`. This solution is definitely bad dependency management because I need to uninstall the existing `protobuf-compiler`, but I was not able to find another simple solution. I am not too familiar with Ubuntu package management. Wouldn't adding the binary path for (v21.5) conflict with the already installed `protobuf-compiler`? 

The exact steps are as follows assuming the EmbeddedProto repo is already in the same directory:
```bash
sudo apt-get install -y python3.8-venv zip unzip g++ autoconf automake libtool curl make

version="21.5"
link="https://github.com/protocolbuffers/protobuf/archive/refs/tags/v$version.zip"
proto_zip_name=protoc-$version
wget $link -O $proto_zip_name.zip
unzip $proto_zip_name.zip
rm $proto_zip_name.zip
proto_lib_name="protobuf-$version"
cd $proto_lib_name/

sudo apt-get remove -y protobuf-compiler
# Build Instructions: https://github.com/protocolbuffers/protobuf/blob/21.x/src/README.md
./autogen.sh
./configure
make -j$(nproc)
sudo make install
sudo ldconfig
cd ../

git submodule update --init --recursive
cd EmbeddedProto
python3 setup.py
cd ../
```

BartHertog, 2023-12-05:
I was resolving #68. I encountered a problem with the warning texts resulting in the text not showing. It would have allowed you to try running even with a version mismatch. The text is now shown correctly and you can continue. You can test this code when selecting the development branch.

Hope this helps.