Embedded-AMS/EmbeddedProto #60: Error executing `python setup.py` on MacOS

juniorsaldanha, 2023-07-05:
I'm facing the following error... 

My environment:
- Python 3.11.4
- pip 23.1.2
- protoc `libprotoc 23.3` installed with brew `brew install protobuf`

```
================================================================================
|                                                                              |
| Three simple things you can do to help improve Embedded Proto:               |
|  * Give private feedback:                                                    |
|        https://EmbeddedProto.com/feedback                                    |
|  * Report an issue in public on Github:                                      |
|        https://github.com/Embedded-AMS/EmbeddedProto/issues                  |
|  * Stay up to date on Embedded Proto via our User mailing list:              |
|        https://EmbeddedProto.com/signup                                      |
|                                                                              |
================================================================================

Checking the version of Python [Success]
Checking your Protoc version [Fail]
Error: 'NoneType' object has no attribute 'group'
```

BartHertog, 2023-07-06:
Hello @juniorsaldanha,

Thank you for writing in. 

Can you help me get this clear? The error you showed was prior to downgrading protobuf. After downgrading, you could not get it to work without adding protobuf to your path. What were the errors you were getting after the downgrade? Is it working now with protoc in your path or not?

Best regards,

Bart

P.S. I am not a MAC user, so I am not able to give you any direct assistance.

juniorsaldanha, 2023-07-06:
@BartHertog Before downgrading, the error was protobuf version. After downgrading to version @21, brew didn't added the protobuf bin to my path, so I had to do it manually. And now its working with protobuf@21 and symlink from `/usr/local/bin/protoc` to `/usr/local/opt/protobuf@21/bin/protoc`. I didn't want to add a new directory to my path, so I choose to just create a symlink on `/usr/local/bin/` which is in my path.

@BartHertog is there a plan to implement the latest version of protobuf, can I help with that? Maybe we can structure the project to support multiple versions from now on.

closed 2023-07-06 by juniorsaldanha