Embedded-AMS/EmbeddedProto #65: Repeated Message Custom Options Not Working in Python Environment

ttchalakov, 2023-11-01:
I am attempting to use the custom options for repeated messages ([source](https://embeddedproto.com/documentation/using-a-message/repeated-fields/)) in python so that I can send data to an Arduino, but I am not able to figure out how to get the custom options to be a visible module in my python 3.8.10 installation. I was not able to find documentation in the website on how to make EmbeddedProto custom options available in my python environment.
Here is some investigation:
I have run the protoc command making sure to include the `-I./generator` option and the resulting python code compiles without issue and all features except the custom options work. The generated python code correctly finds the embedded_proto_options.proto file. However when I run the python code, it is not able to find `embedded_proto_options_pb2` as it expects it to be a module.
```bash
test/test_example.py:16: in <module>
    from motor_driver.generated import example_pb2
motor_driver/generated/example_pb2.py:14: in <module>
    import embedded_proto_options_pb2 as embedded__proto__options__pb2
E   ModuleNotFoundError: No module named 'embedded_proto_options_pb2'
```
To work around this, I made a module named 'embedded_proto_options_pb2' and copied the `./generator/embedded_proto_options_pb2.py` file into that module. Now when I run my code, the module is found and 'embedded__proto__options__pb2' is accessible. However I get the following error:
```bash
test/test_example.py:16: in <module>
    from motor_driver.generated import example_pb2
motor_driver/generated/example_pb2.py:17: in <module>
    DESCRIPTOR = _descriptor_pool.Default().AddSerializedFile(b'\n\rexample.proto\x1a\x1c\x65mbedded_proto_options.proto\"\[rest omitted for purpose of this git issue]')
E   TypeError: Couldn't build proto file into descriptor pool: Depends on file 'embedded_proto_options.proto', but it has not been loaded
```
I am not sure how to load the proto file. Any suggestions to make using EmbeddedProto with python more seamless would be great.

BartHertog, 2023-11-04:
I am afraid that this is not possible without effort. Custom options are specific to the implementation of the code generator. The embedded_proto_options file is for the protobuf generator itself. If you would like to use it you would have to modify the Python code generator in protoc.

For repeated fields in Python you will have to remember to send not too many items to your microcontroller.

BartHertog, 2023-11-05:
I forgot to type that I will make this more apparent on the Embedded Proto website documentation.

closed 2023-12-05 by BartHertog