Embedded-AMS/EmbeddedProto #70: ImportError embedded_proto_options_pb2

Fabien-B, 2024-01-17:
Hello,
I updated EmbeddedProto from 3.3.3 to 3.4.2 with the latest protobuf version (25.2), and it crashes complaining about not finding `embedded_proto_options_pb2`.

I managed to resolve the issue by generating the file with:
`protoc -I generator --python_out=generator/EmbeddedProto embedded_proto_options.proto`

However this is not written in the documentation. Is it necessary? Could it be done in the setup.py so we don't have to manually generate it?

I did not changed my .proto file.

BartHertog, 2024-01-18:
Yes, I have seen this and with me as well it appeared and disappeared. The proto file needs to be generated during the setup process but for some reason, it does not. 

What you could do is try to download EmbeddedProto in a different folder but on the troubled machine. If the setup works there it has to do with artifacts in the original folder. If that is the case try running `python setup.py --clean` to remove artifacts from the previous setup.

BartHertog, 2024-01-29:
Hi Fabien,
How is it going at this moment? Are you able to use Embedded Proto?

closed 2024-02-06 by BartHertog