Embedded-AMS/EmbeddedProto #72: ImportError: cannot import name 'embedded_proto_options_pb2' from 'EmbeddedProto'

LiangGuoYu, 2024-03-11:
Hello!
When I run this command it gives me the following error. What should I do?

Step1:
EmbeddedProto# python setup.py

================================================================================
|                                                                              |
| Three simple things you can do to help improve Embedded Proto:               |
|  * Give private feedback:                                                    |
|        https://EmbeddedProto.com/feedback                                    |
|  * Report an issue in public on Github:                                      |
|        https://github.com/Embedded-AMS/EmbeddedProto/issues                  |
|  * Stay up to date on Embedded Proto via our User mailing list:              |
|        https://EmbeddedProto.com/signup                                      |
|                                                                              |
================================================================================

Checking your Protoc version [Success]
Creating a virtual environment for Embedded Proto. [Success]
Installing EmbeddedProto in the virtual environment. [Success]
Setup completed with success!

Step2:
EmbeddedProto# python setup.py --include /usr/local/include/google/protobuf/

================================================================================
|                                                                              |
| Three simple things you can do to help improve Embedded Proto:               |
|  * Give private feedback:                                                    |
|        https://EmbeddedProto.com/feedback                                    |
|  * Report an issue in public on Github:                                      |
|        https://github.com/Embedded-AMS/EmbeddedProto/issues                  |
|  * Stay up to date on Embedded Proto via our User mailing list:              |
|        https://EmbeddedProto.com/signup                                      |
|                                                                              |
================================================================================

Checking your Protoc version [Success]
Creating a virtual environment for Embedded Proto. [Success]
Installing EmbeddedProto in the virtual environment. [Success]
Setup completed with success!

EmbeddedProto# protoc --plugin=protoc-gen-eams -I/home/my_proto --eams_out=./generator_cpp embedd_test.proto
Traceback (most recent call last):
  File "./venv/bin/protoc-gen-eams", line 33, in <module>
    sys.exit(load_entry_point('EmbeddedProto', 'console_scripts', 'protoc-gen-eams')())
  File "./venv/bin/protoc-gen-eams", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.8/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/EmbeddedProto/generator/EmbeddedProto/main.py", line 33, in <module>
    from EmbeddedProto.ProtoFile import ProtoFile
  File "/home/EmbeddedProto/generator/EmbeddedProto/ProtoFile.py", line 31, in <module>
    from .TypeDefinitions import *
  File "/home/EmbeddedProto/generator/EmbeddedProto/TypeDefinitions.py", line 31, in <module>
    from .Field import Field
  File "/home/EmbeddedProto/generator/EmbeddedProto/Field.py", line 32, in <module>
    from . import embedded_proto_options_pb2
**ImportError: cannot import name 'embedded_proto_options_pb2' from 'EmbeddedProto'** (/home/EmbeddedProto/generator/EmbeddedProto/__init__.py)
--eams_out: protoc-gen-eams: Plugin failed with status code 1.

BartHertog, 2024-03-11:
Hello @LiangGuoYu ,

A possible solution could be rerunning the setup.py script with the --clean option.

Best regards,

Bart

LiangGuoYu, 2024-03-12:
Thanks, but I've already done that before and it gives the same error。

> Hello @LiangGuoYu ,
> 
> A possible solution could be rerunning the setup.py script with the --clean option.
> 
> Best regards,
> 
> Bart

Thanks, I've tried the python setup.py --clean command, but it still gives the same error。

root@ubuntu:/home/liang/Documents/test_protobuf/EmbeddedProto# python --version
Python 3.8.10
root@ubuntu:/home/liang/Documents/test_protobuf/EmbeddedProto# which python
/usr/bin/python
root@ubuntu:/home/liang/Documents/test_protobuf/EmbeddedProto# protoc --version
libprotoc 25.1
root@ubuntu:/home/liang/Documents/test_protobuf/EmbeddedProto# which protoc
/usr/local/bin/protoc

root@ubuntu:/home/liang/Documents/test_protobuf/EmbeddedProto# python setup.py --clean
```text
================================================================================
|                                                                              |
| Three simple things you can do to help improve Embedded Proto:               |
|  * Give private feedback:                                                    |
|        https://EmbeddedProto.com/feedback                                    |
|  * Report an issue in public on Github:                                      |
|        https://github.com/Embedded-AMS/EmbeddedProto/issues                  |
|  * Stay up to date on Embedded Proto via our User mailing list:              |
|        https://EmbeddedProto.com/signup                                      |
|                                                                              |
================================================================================
```
Cleaning the folder.
Checking your Protoc version [Success]
Creating a virtual environment for Embedded Proto. [Success]
Installing EmbeddedProto in the virtual environment. [Success]
Setup completed with success!
root@ubuntu:/home/liang/Documents/test_protobuf/EmbeddedProto# protoc --plugin=protoc-gen-eams -I/home/liang/Documents/test_protobuf/my_proto --eams_out=./cpp embedd_test.proto
Traceback (most recent call last):
  File "./venv/bin/protoc-gen-eams", line 33, in <module>
    sys.exit(load_entry_point('EmbeddedProto', 'console_scripts', 'protoc-gen-eams')())
  File "./venv/bin/protoc-gen-eams", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.8/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "\<frozen importlib._bootstrap\>", line 1014, in _gcd_import
  File "\<frozen importlib._bootstrap\>", line 991, in _find_and_load
  File "\<frozen importlib._bootstrap\>", line 975, in _find_and_load_unlocked
  File "\<frozen importlib._bootstrap\>", line 671, in _load_unlocked
  File "\<frozen importlib._bootstrap_external\>", line 848, in exec_module
  File "\<frozen importlib._bootstrap\>", line 219, in _call_with_frames_removed
  File "/home/liang/Documents/test_protobuf/EmbeddedProto/generator/EmbeddedProto/main.py", line 33, in <module>
    from EmbeddedProto.ProtoFile import ProtoFile
  File "/home/liang/Documents/test_protobuf/EmbeddedProto/generator/EmbeddedProto/ProtoFile.py", line 31, in <module>
    from .TypeDefinitions import *
  File "/home/liang/Documents/test_protobuf/EmbeddedProto/generator/EmbeddedProto/TypeDefinitions.py", line 31, in <module>
    from .Field import Field
  File "/home/liang/Documents/test_protobuf/EmbeddedProto/generator/EmbeddedProto/Field.py", line 32, in <module>
    from . import embedded_proto_options_pb2
ImportError: cannot import name 'embedded_proto_options_pb2' from 'EmbeddedProto' (/home/liang/Documents/test_protobuf/EmbeddedProto/generator/EmbeddedProto/__init__.py)
--eams_out: protoc-gen-eams: Plugin failed with status code 1.
root@ubuntu:/home/liang/Documents/test_protobuf/EmbeddedProto#

BartHertog, 2024-03-12:
How unfortunate,
Can you check the folder `EmbeddedProto/generator/EmbeddedProto`? Does it contain the file `embedded_proto_options_pb2.py`? 
If not, let's try the manual workaround by running the following: 
```protoc -I . --python_out=EmbeddedProto embedded_proto_options.proto```
You have to run this command from the generator folder. 
After completion the `embedded_proto_options_pb2.py` should be present.

LiangGuoYu, 2024-03-13:
> How unfortunate, Can you check the folder `EmbeddedProto/generator/EmbeddedProto`? Does it contain the file `embedded_proto_options_pb2.py`? If not, let's try the manual workaround by running the [...]

Thank you very much！It works！

closed 2024-03-13 by LiangGuoYu

mateusz-, 2024-08-16:
I just ran into this issue as well and the workaround worked for me.