Embedded-AMS/EmbeddedProto #76: EmbeddedProto v3.5.0 fails with Protocol Buffers v27.1

EthanG78, 2024-06-10:
As the title described, EmbeddedProto v3.5.0 fails with the latest Protocol Buffers release, v27.1. Failure does not occur when running `setup.py`, it occurs when trying to run the protoc compiler.

**NOTE**: I understand that it is clearly advertised that EmbeddedProto is only tested on Protocol Buffers v25.1, I just wanted to put this on the dev team's radar.

Included in the code snippet below is the .proto file I was trying to compile:
```proto
syntax = "proto3";

import "google/protobuf/timestamp.proto";

enum CommandType { 
  COMMAND_UNSPECIFIED = 0; 
  COMMAND_START = 1;  
  COMMAND_DATA = 2; 
} 

message Packet {
  float frequency = 1;  
  float sample_frequency = 2;  
  uint32 num_samples = 3;
  repeated uint32 dac_samples = 4 [packed=true];  
  repeated uint32 adc_samples = 5 [packed=true]; 
}

message Command { 
  CommandType command_type = 1;
  optional uint32 num_frequencies = 2; 
  repeated float frequencies = 3 [packed=true];
  optional Packet data_packet = 4;
  google.protobuf.Timestamp last_updated = 5;
} 
```

This is the command that was used to try to compile the file (with paths omitted):
```bash
protoc --plugin=protoc-gen-eams=<path_to_embeddedproto>\protoc-gen-eams.bat -I=.\protos\ --eams_out=.<output_directory> <file_name>.proto
```

These commands were executed in a Windows 11 Home environment running Python v3.10.6 and Git version 2.34.0.windows.1.

Upon installing Protocol Buffers v25.1 and running the same command on the same .proto file, the compiler successfully generated the output file.

BartHertog, 2024-06-10:
Hello @EthanG78,

Thank you for writing in! This most certainly helps. 

I thought it was just a simple version update of the internal config file that started working. Unfortunately, some things changed in the protoc compiler, so it will take some more time. Unfortunately, I now broke the build. I will look at this again tomorrow.

Will keep you posted.

Best regards,
Bart

BartHertog, 2024-06-11:
Oke, this is not a straightforward problem. 

Just rubber ducking here:

So protobuf has this file "descriptor.proto" which is used to communicate custom features between the .proto file and the plugin, the EmbeddedProto generator in this case. 

With the new version of protobuf (v27) the descriptor.proto file was changed. 

It now has a new nested message definition like this:
```proto
message A
{
  message B
  {
  
  }
}

message C
{
 A::B b;
}
```
Now using B in C is in itself not a problem where it is not that message A also has repeated and string fields, making it a template class. This results in a problem that so far has not been solved in the generated. Namely deducting the exact type for `A<???>:B`.

Maybe generating the descriptor.proto or the embedded_proto_options.proto file which uses it can be skipped.

BartHertog, 2024-06-12:
The issue has been resolved. 
It turned out that generating the files was already skipped. What was done was including descriptor.proto for test purposes, to see if items (well-known types) could be included from the files included with protoc. Selecting a different file for that purpose solved the build problem.

Thanks again for your report. 

Best regards,

Bart

closed 2024-06-12 by BartHertog

EthanG78, 2024-06-12:
Awesome, sounds good. Thank you!