Embedded-AMS/EmbeddedProto #78: Protobuf Version Mismatch

mateusz-, 2024-08-16:
Hello,

The [website installation instructions](https://embeddedproto.com/documentation/installation/) state the following:

```
What is required to be able to generate source files based on .proto files:

Python 3.8 and up
Pip
Protobuf v21.5 and up (tested with v25.1)
Git
```

Per instructions above, I downloaded Protobuf v25.1 and proceeded with the installation:

```
$ PATH=$PATH:../../Utilities/protoc/bin python setup.py

================================================================================
|                                                                              |
| Three simple things you can do to help improve Embedded Proto:               |
|  * Give private feedback:                                                    |
|        https://EmbeddedProto.com/feedback                                    |
|  * Report an issue in public on Github:                                      |
|        https://github.com/Embedded-AMS/EmbeddedProto/issues                  |
|  * Stay up to date on Embedded Proto via our User mailing list:              |
|        https://EmbeddedProto.com/signup                                      |
|                                                                              |
================================================================================

Checking your Protoc version [Caution]

The version of Protoc (v25.1) you have installed is not the same as the version of
the protobuf python package (v27.1).These are your options:
        1. Install a matching version of Protoc.
        2. Change the version of Embedded Proto.
        3. Ignore the difference and try if it works for you (please let us know if it does not stating the versions you used).

Ignore the difference [Y/n]: 2
Ignore the difference [Y/n]: 2
Ignore the difference [Y/n]: n
Stopping the setup.
```

It appears that updating to v27.1 fixes the problem:

```
PATH=$PATH:../../Utilities/protoc/bin python setup.py

================================================================================
|                                                                              |
| Three simple things you can do to help improve Embedded Proto:               |
|  * Give private feedback:                                                    |
|        https://EmbeddedProto.com/feedback                                    |
|  * Report an issue in public on Github:                                      |
|        https://github.com/Embedded-AMS/EmbeddedProto/issues                  |
|  * Stay up to date on Embedded Proto via our User mailing list:              |
|        https://EmbeddedProto.com/signup                                      |
|                                                                              |
================================================================================

Checking your Protoc version [Success]
Creating a virtual environment for Embedded Proto. [Success]
Installing EmbeddedProto in the virtual environment. [Success]
Setup completed with success!
```

Also, the earlier prompt is confusing. It provides various options (1, 2, 3) but then asks for a yes or no.

BartHertog, 2024-08-16:
Hello @mateusz-,

Thank you for trying Embedded Proto and writing in. What you describe is different from what you are supposed to encounter. By reporting such issues, you help improve Embedded Proto. I am grateful for that feedback.

At first glance, not all documentation and code were updated when the protobuf version was incremented to 27.1. The readme file was partially updated when the Protobuf version was updated, but the website was not. 

This is what I will do now:
1. Check whether the above statement is indeed the problem.
2. If so, update all locations where the protobuf version was mentioned.
3. Make a checklist with all points that should be updated before releasing a version with a protobuf update.

Regarding your second remark, I understand what you mean about the confusing prompt. You are given three options, and next, you are asked for a y/n answer. That does not match. I have written a new prompt below. Is that text clear?

```
The version of Protoc (vYY) you installed is different from the protobuf python package (vXX) version, which comes with the version of Embedded Proto you installed. You could ignore the difference and try to run anyway or perform any of these manual actions:
        1. Install a matching version of Protoc.
        2. Change the version of Embedded Proto.

Ignore the difference [Y/n]:
```

Best regards,

Bart

BartHertog, 2024-08-16:
I have a bit of a brain dump regarding the second remark. Looking at the code, it is not as simple as just changing the text. The third option only appears when a version higher than 21 is used. Below that, the installation is, by default, a fail. Version 21 introduced breaking changes. This guards against that. 

Nonetheless, I will make the text and opinions given clear.

BartHertog, 2024-08-16:
This is what it has become:
```
The version of Protoc (vX.Y)  you installed is different from the Protobuf Python package version (vU.Z). These are your options:
  1. Manually install a matching version of Protoc.
  2. Manually change the version of Embedded Proto.
  3. Ignore the difference and try if it works for you (please let us know if it does not stating the versions you used).
  
  Would you like to try option 3? [Y/n]: 
```

BartHertog, 2024-08-16:
The changes can be found in the development branch. I also changed the installation page on the website to refer back to the readme file to have fewer places mentioning a specific version.

mateusz-, 2024-08-19:
Thank you, Bart.

The new prompt makes sense to me.

I'll close the issue.

closed 2024-08-19 by mateusz-