Embedded-AMS/EmbeddedProto #81: Unclear license conditions

andrey-pr, 2025-01-20:
> You can use Embedded Proto for free in open source projects or for testing. However, on demand support is not available only if you have a commercial license. For open source projects, you can download the source code from Github. The code is licensed under the GNU General Public License V3.0, which you can use for all your non-commercial projects.

Is this library allowed to use commercially with open source code? According to GPL-3.0, its possible, but according to description of dual-licensing model in readme and on website not

BartHertog, 2025-01-22:
Hello @andrey-pr,

Thank you for your interest in Embedded Proto!

The license is like a discussion tree. The main question is: Do you use the code commercially or not? If you use it commercially, the GPL license does not apply. 

I hope this clarifies your question. Feel free to ask me more if desired.

Best regards,

Bart

closed 2025-02-10 by BartHertog

amari-ca, 2026-09-18:
Offering something for non-commercial use only under GPL 3.0 is inherently contradictory.

From the GPL 3.0 license:

> All other non-permissive additional terms are considered “further restrictions” within the meaning of section 10. If the Program as you received it, or any part of it, contains a notice stating that it is governed by this License along with a term that is a further restriction, you may remove that term.

By offering this code with a GPL 3.0 license non-commercially, you are giving users permission to ignore the non-commercial term, and use it commercially anyways as long as they follow GPL 3.0.

BartHertog, 2026-09-20:
Hello @amari-ca,

You are right, and thank you for pointing it out. The GPLv3 does not allow a non-commercial restriction, and we do not intend one. The earlier wording, including my answer above, was imprecise.

The actual model is this. Under the GPLv3 you can use Embedded Proto free of charge in any project, commercial or not, as long as the product you distribute complies with the GPLv3. That includes the generated code, which embeds code from the templates in this repository. The commercial license exists for products whose source code is not published. Internal evaluation and testing before purchase are free either way.

The README, LICENSE file and website are being updated to say exactly this. The word "non-commercial" is gone from them.

Best regards,

Bart