Embedded-AMS/EmbeddedProto #83: `mutable_*` not available for enum fields

tokoro10g, 2025-09-18:
Hi Bart,

I realized that the `mutable_*` methods are not available only for enum fields. Is there any reason behind it, or is it simply missing?

Thanks.

BartHertog, 2025-09-20:
I had to check this because I was unsure from what I recalled from the top of my head. I tried to mimic the Google C++ interface as much as possible, and what I recalled is that an enum did not have the `mutable_*` function. 

So, I generated the default C++ code for `test\proto\simple_types.proto` test file with protoc. In the header file, I am not able to find the `mutable_*` function for the enums used in the messages in that file.

So yes, I would expect it as well, but have not added it on purpose.

BartHertog, 2026-09-11:
Hello @tokoro10g,

The upcoming v4 release includes the mutable_* version for enums with commit 763f5e3eb51ae35c5977a503caae8c47bab75e39.
I will push the changes to the develop branch either today or early this week.

Best regards,

Bart

closed 2026-09-11 by BartHertog

tokoro10g, 2026-09-19:
That's a great news! Thank you!